chillibasket/walle-replica #8: ESP8266 with motor shield

Eisbaeeer, 2019-11-29:
Hi
Great project. Did you think about to control the wall-e with an esp8266 with motor shield?
There is also an esp32 cam available.
Let´s see what I can do ;-)

chillibasket, 2019-11-29:
I haven't had the chance to try out the ESP8266 or ESP32 yet, but it definitely could be a substitute to the Arduino & Raspberry Pi I used. Maybe you could also include an audio shield to add sound effects? I look forward to seeing your approach!

Eisbaeeer, 2019-11-30:
I programmed a very simular software to control an other educational robot. You will find it on www.kidbuild.de
The only thing to do is to add the additional servos and the sound module. For sound I will use a small mp3 player with micro-sd card. The board is able to drive a speaker directly with 2 watts. It´s called dfplayer. If you agree, I can do a pull-request, if the program is ready for public.

cavamora, 2019-12-16:
Hi! I am also doing the same thing! Trying to use an esp8266 to control the robot. Thinking about using mongoose os also. With blynk as remote control.

By the way, did you managed to run the project, asis, in the esp8266? In here, the serial info in the monitor is messy. Already checked the serial port speed...

Eisbaeeer, 2019-12-17:
@cavamora I will use 2 controllers who will build an internal wifi network. The first is the esp32-cam used for live video distribution, the second is the esp8266 with motor shield to control the Wall-E. The esp32-cam is working in AP mode, the esp8266 as client to the esp32-cam. I don't need any app to control the Wall-E. You are able to connect to the wifi and then call the embedded webpage on the esp. The page will transmit the live picture and buttons to control the Wall-E. Actual state. Movement is working, live video is working, sounds via DF-player is working. Pending is the code for the i2c servo board. I will push my solution on my page soon. Www.kidbuild.de

cavamora, 2019-12-17:
great!!!! 
in my case, I am not using a camera. just one esp8266, connected to my wifi and to Blynk cloud. receiving the commands from RESTAPI or from BLYNK cloud, using Mongoose OS SDK instead of Arduino SDK.
i intend to integrate with ALEXA/Google Assistant to send commands to wall-e by voice.

Status:
Already ported the SERVOS(PCA9685) and SERIAL (just for testing) source code to Mongoose OS project.
Blynk integration OK.

Pending:
Controll DC 12v Motors (using 4 digital pins from esp8266 and 2 PWM pins  on PCA9685)
Move SERIAL protocol to a RESTAPI
MP3 player (DFPlayer) using serial UART for controlling
Find a way to recharge a 12v battery

For controlling the DC 12v Motors, as I am using L298N, I pretend to MIX it with the PCA9685 (for PWM commands). Using just 4 digital pins from ESP8266.  Also, will need to change the functions a little, as the Pins are a bit different (there is no break pin...)

I suggest we exchange emails to keep the topic: gustavocmorais@gmail.com

Cheers

Eisbaeeer, 2019-12-18:
Good hint. Don´t spam Simon´s git. I created a blog with comment section. Please post there, so we can discuss solutions. You will find it there. [http://www.weimars.net/?page_id=362](url)
Thank´s to Simon for this project!

closed 2019-12-18 by Eisbaeeer