chillibasket/walle-replica #9: wall-e upgrade requires additional code for enhancements to work (new moving model parts available)

ciorik, 2020-01-21:
1. original wall-e can be upgraded to have movable eyebrows as seen here: https://www.thingiverse.com/thing:3955562
https://www.bilibili.com/video/av74553688?from=search&seid=13367369835271189022
2. two additional MG90S servo motors can be added to upgraded wall-e for each eyebrow
3. Additional code is required in order to control the eyebrows

chillibasket, 2020-01-22:
Adding the two extra servo motors to the Arduino code is relatively straight-forward; it's just a matter of adding two extra items into each of the motor-control arrays. However, I'm not quite sure how the eyebrows should be controlled from the web interface. Do you think three simple on/off buttons with which you can activate the left, right, or both eyebrows is enough?  

Overall, since this is an additional upgrade to the basic Wall-E design, I propose to keep this version of the code in a separate branch to the main version in the repository.

ciorik, 2020-01-22:
that would be great

ciorik, 2020-01-26:
I just finished the wall-e assembly with all servo motors in place. It seems that the eyebrows servos needs to rotate ~5-10 degrees counterclock direction in order to open then 5-10 degrees clock direction to close. See my screenshot: https://prnt.sc/qt0ezp

chillibasket, 2020-01-26:
Looks good, you will be able to fine-tune the exact servo positions with the servo calibration sketch once I add the eyebrow code. I'm currently doing some major upgrades on the web-interface (for example I'm adding xbox and playstation controller support!), so it may be another 1-2 weeks until I get around to adding the eyebrow support.

chillibasket, 2020-02-23:
I've added a branch called "eyebrows" to the GitHub repository: https://github.com/chillibasket/walle-replica/tree/eyebrows

The code within that branch should have all of the changes required to support the eyebrows. To clone the branch from the Raspberry Pi, you can use the following command: `git clone -branch eyebrows https://github.com/chillibasket/walle-replica.git`. Make sure that you run the servo calibration sketch first so that you can get the correct minimum (eyebrows down) and maximum (eyebrows up) positions for the eyebrow servos. 

Since I don't have any eyebrows on my own robot, I'm not able to test if the code actually works. Please let me know if you come across any issues or bugs which need to be fixed.

closed 2020-02-23 by chillibasket