chillibasket/walle-replica #13: Web interface direction issue

kesenls, 2020-01-30:
Hello and thank you for you great code.
One small issue . Allthough motors are moving correctly back and forth , the left and right direction are opposite.

When i move the stick to the right the bot goes left
I checked all the cabling no issue there

Mybe add a switch to mirror left and right direction ?

Thank you in advance

Im also checking the posibility to have obstacle avoid using web cam and opencv and a botton for the robot to randomly move arround autonomoysly ?

chillibasket, 2020-01-30:
The easiest way to fix the problem is to modify the Arduino code on line 230 by adding a minus sign to the 2.55 value:
`if (firstChar == 'X' && number >= -100 && number <= 100) turnVal = int(number * -2.55);`

If this doesn't solve your problem, then the motors are wired backwards. I just noticed now that the X and Y labelling in the comments is wrong. X refers to left/right turning, while Y is fowards and backwards. Anyway, I'll double check the wiring diagram and code to see if direction really is inverted, and fix it in the next update.

I'm not planning on adding obstacle avoidance and random movement. Recognising and tracking objects is definitely possible using the Raspberry Pi camera, but obstacle avoidance is more difficult since the camera is not able to sense depth. If you are interested in learning about computer vision, this tutorial shows some of the basics: https://www.youtube.com/watch?v=tpwokAPiqfs&t=11s

closed 2020-01-30 by chillibasket

kesenls, 2020-01-30:
hello and thank you for your fast reply

i dont think its an arduino issue since when i send a or d from serial on
arduino direction works as it supposed . ill try the code though and come
back to you

kesenls, 2020-01-30:
yeap that fix did the trick thank you !

kesenls, 2020-01-31:
Last suggestion is would be great to create a facebook group where we could
all upload our walles and Inspire more to do so and also exchange code and
ideas.
For example i went bigger and already altered many parts to fit standard
servos

Στις Πέμ, 30 Ιαν 2020, 9:08 μ.μ. ο χρήστης Unlim ited <kesenlis@gmail.com>
έγραψε:

chillibasket, 2020-01-31:
There already is a Facebook group for people building Wall-E called: **WALL-E Builders Club**. On the group there are many people building full-size robot, as well as others building smaller scale models such as my design.