chillibasket/walle-replica #16: Corrections to instructions

starlord1980, 2020-09-17:
Just a note that I had to do a few things differently than the guide to get the webserver going, I noticed it was coded in python3, I had to use:

_sudo apt-get install python3-pip_
and then
_sudo pip3 install flask_
  as opposed to what was written in the guide otherwise I got a 'Module not found' error.

Also then had to install git via
_sudo apt install git_

Also _sudo pip3 install pygame_ wasnt in there but I figured that out.

chillibasket, 2020-09-17:
Yes I think you are right about the python3 commands. The instructions have been up for over a year, so I'm surprised no one else said they had a problem with that... 

Regarding the "git", "pygame" and "libsdl-mixer" packages, they were included by default in the past versions of Raspbian so I left them out of the instructions. I guess there's no harm explicitly putting the install commands into the instructions.

What OS version are you using with your RPi?

starlord1980, 2020-09-17:
Hi, thanks for getting in touch I'm working it all out at the moment, I'm only learning python at the moment so its not completely over my head but I'm getting there. Just googling the errors until I get it working.
Its the latest Raspberry Pi OS 32bit (CLI only) I'm running on a Pi ZeroW 1.1. Planning on using a microUSB>USB-B cable to connect to the arduino.

chillibasket, 2020-09-17:
I just had a quick look at the Raspberry Pi OS "Buster" info and it seems that the "Lite" version does not include these packages by default (causing the errors you saw). They are installed in the "Full" version.

Anyway, I have now updated the instructions to use python3 commands. I've also added a line with the commands to install those additional packages.

closed 2020-09-17 by chillibasket