chillibasket/walle-replica #17: How to get power to the rapsberry pi ?

zannkukai, 2021-01-09:
Hi All,
Thanks you for this great project. At this time, I tried to wiring a the component together and place them into wall-e.
I would like to know how to power the raspberry pi when this one are place into wall-e.
It's sure I can't use my micro usb power supply with the cable following the robot ;-)

I have a battery 12v/550mAh for this project. It work well with arduino, but when I plug the USB cable to connect Arduino to rPi , the RPi doesn't turn on (rPi v3 model B). It seems rPi need more power (2,5A ?) to be turn on. 
What's your solution/proposition to fix this issue ? 

Thanks a lot

chillibasket, 2021-01-09:
The USB cable between the Raspberry Pi and Arduino does not supply any power. The RPi needs to be powered separately via its dedicated micro-usb power connector. In the parts list on Thingiverse and my website, I've linked a 12V to 5V buck converter which also has a USB power output port which can be connected directly to the Raspberry Pi. If you are using a different buck converter, you may need to cut off one end of a USB cable so that you can solder it to the converter. Yes, the RPi needs quite a bit of power but it seems to be working fine with the 5A 12V to 5V buck converter I am using (this converter also powers my servo motors).

zannkukai, 2021-01-09:
I'm also used the 12v/5v buck converter. I'm using this one : [https://fr.aliexpress.com/item/32890189436.html?spm=a2g0s.9042311.0.0.27426c37Wfg8PT&af=d99%3A&aff_platform=api-new-link-generate&sk=_bWqjGRD&aff_trace_key=0b8a79fe455a47b78022b093076d4cd1-1610205617536-00564-_bWqjGRD&cn=32890189436&dp=91.176.103.251-&terminal_id=6d04e58ef6d74c25958aad6a7af7e507&tmLog=new_Detail](url).

I didn't think about converter using the USB output.... sometimes, I forget to think about the easiest solution.
I will test ASAP when my battery will be full charged.
About your battery, I used a 8800mAh 12V battery. For me it's ok for the project, can you confirm ? Maybe it will be useful to specify this specification on your site ?

chillibasket, 2021-01-09:
I would recommend a capacity of around 2100mAh; this battery size is small enough to fit nicely inside the robot and should be enough to power Wall-E for an hour. Batteries of other capacities will work fine electronically, but you might have difficulty fitting the 8800mAh battery inside the body. I've added the details to my website now.

zannkukai, 2021-01-10:
thanks for all theses informations and many mant thanks for this wonderfull project !

closed 2021-01-10 by zannkukai