chillibasket/walle-replica #18: Support for oLED display

hpkevertje, 2021-03-14:
I've modified your code to support an 1.3 inch SH1106 based oLED display using the u8g2 library. 
If you're interested I'm happy to share it here. Would that be OK?

chillibasket, 2021-03-15:
Hi, if you make the code public on your GitHub profile I can add a link to it on my website and in the Readme so that other makers can find it.

hpkevertje, 2021-03-17:
I've added it to https://github.com/hpkevertje/wall-e
I'm not used to creating projects in GitHub and comitting code, so if I did something incorrect, please point it out so I can correct.

chillibasket, 2021-03-20:
The code looks good and is nice and simple! The only thing missing is a short explanation of how to wire up the display to the Arduino. The way you have committed your GitHub code is fine as a stand-alone project, however when modifying/extending an existing project people usually do the following:

  - "[Fork](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/working-with-forks)" the project first (this creates a copy of entire repository on your own account).
  - Then configure the fork so that it is synced to the upstream repository (this would be my repository).
  - Then they can edit and commit the code in their fork as much as they like.
  - If you would then want me to consider integrating the updated code into my master repository, you would create a "[Pull Request](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork)".

It's quite confusing to people who are new to the process and it took me a while to get my head around it! So the question is whether we should add this code to my Master repository or not...

chillibasket, 2021-03-20:
Anyway, I'll add some links to your code this weekend.

hpkevertje, 2021-03-21:
Thanks Simon,

I'll update the documentation with a wiring diagram and I'll fork your repository properly. The 4 steps you describe give me enough guidance to find my way. You should see a 'pull request' somewhere next week.

hpkevertje, 2021-03-21:
So I forked, synced my fork to yours, commited my code, including an updated readme file with instructions and created the pull request.

Again, I'm new with github, so if I did something wrong, feel free to point it out so I can correct.

chillibasket, 2021-03-21:
Everything looks good... could you just make two minor changes and then I am happy to merge it.
1.  Move the line `#include <U8g2lib.h>` down to the object instantiation like this:
  ```cpp
  #ifdef OLED
    #include <U8g2lib.h>
    U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, 10);
  #endif
  ```
  This makes it so that the code will still compile if the user doesn't have the u8g library installed and isn't using the oLed code.

2. In the Readme beside the "oLed Display" title add a link to your GitHub profile: `oLed Display (Contributed by: [hpkevertje](https://github.com/hpkevertje))`

hpkevertje, 2021-03-22:
I've made those modifications and they are committed. Thanks for the feedback and merging.

closed 2021-03-22 by chillibasket

chillibasket, 2021-03-22:
Ok it has now been merged. Thanks for sharing the code; I know this is a feature many other makers have been looking for.