chillibasket/walle-replica #20: fallo para calibrar

davrodbla, 2021-04-09:
cuando intento calibrar los sevomotores comienzan a girar sin parar, no responden a los comandos 'a' 'd' 'c' 'z', solo paran de girar cuando envio el comando 'n'.

chillibasket, 2021-04-10:
If the letter "n" works then the other commands should also work; pressing the other letter won't display anything on the screen but they do send a command to the servo motors. Most likely the problem is related to hardware and not the software, which causes there to be no movement when you send the commands. Here are some things you should check:

- Are the servo motors getting enough power? Test the pins with a multimeter to check if they are getting 5V.
- Is the Arduino able to communicate with the servo controller board? The voltage of the signal pins going to the servo motor should change as when it command the servo to move.
- Do the servo motors work when you use the sample code for the servo controller board: [https://learn.adafruit.com/16-channel-pwm-servo-driver/using-the-adafruit-library](https://learn.adafruit.com/16-channel-pwm-servo-driver/using-the-adafruit-library)

davrodbla, 2021-04-11:
movimiento si hay por lo que descarto tema de tensiones, el problema es que en el momento que abro el monitor serie el primer servo comienza a girar continuamente sin que yo le envío ningún comando de movimiento y solo para cuando envío el comando 'n' y empieza a girar en el sentido contrario y de nuevo no para hasta que envío el comando 'n', y así un servo detrás de otro. he revisado los tres parámetros que me comentas y están correctos

chillibasket, 2021-04-11:
Ok it sounds as if the code and wiring is working fine, but you may have received the wrong type of servo motor. Do the servo motors rotate more than 180°? Normal servo motors do not keep spinning and can only rotate up to 180° before they stop. You may have received "continuous rotation" servo motors which will keep spinning but their position cannot be precisely controlled, making them unsuitable for this application.

davrodbla, 2021-04-11:
Sí, mis motores son rotatorios sin paradas, supongo que ese será el fallo. Pediré nuevos y lo intentaré otra vez, muchas gracias por todo.
Un saludo, David

Obtener Outlook para iOS<https://aka.ms/o0ukef>

closed 2021-04-11 by chillibasket