chillibasket/walle-replica #29: Autostart not working

myMartek, 2023-11-22:
Hi,

I have installed the Webinterface on my RaspberryPI 4 with 64bit Raspberry PI OS.

I also added the startup script to automatically start the webinterface but unfortunately the web interface will not autostart.

The error message I got from journalctl is:

`Nov 22 20:50:57 walle walle[4494]: pygame 2.1.2 (SDL 2.26.5, Python 3.11.2)
Nov 22 20:50:57 walle walle[4494]: Hello from the pygame community. https://www.pygame.org/contribute.html
Nov 22 20:50:57 walle walle[4494]: Traceback (most recent call last):
Nov 22 20:50:57 walle walle[4494]:   File "/home/pi/walle-replica/web_interface/app.py", line 37, in <module>
Nov 22 20:50:57 walle walle[4494]:     pygame.mixer.init()
Nov 22 20:50:57 walle walle[4494]: pygame.error: ALSA: Couldn't open audio device: Unknown error 524
Nov 22 20:50:57 walle systemd[1]: walle.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 20:50:57 walle systemd[1]: walle.service: Failed with result 'exit-code'.`

I found out that the same error comes whenever I run the script with sudo or as another user for example. I have absolutely no clue why it is not working on startup. Do you have any idea?

dkrey, 2024-01-28:
You need to add this environment variable to your systemd service
Environment=XDG_RUNTIME_DIR=/run/user/1000

This is my working unit file.

```
[Unit]
Description=Start Wall-E Web Interface
After=network.target

[Service]
Type=simple
WorkingDirectory=/home/pi/walle-replica/web_interface
Environment=XDG_RUNTIME_DIR=/run/user/1000
ExecStart=/usr/bin/python3 app.py
Restart=always
SyslogIdentifier=walle
User=pi
Group=pi
RestartSec = 5
KillMode = process
SendSIGHUP = no

[Install]
WantedBy=multi-user.target

w45354, 2024-02-06:
I applied the autostart settings. it can autostart but only if a monitor is connected. Has anyone met this issue before?

TheBanshee71, 2024-03-06:
im having the same problem. Any ideas? IM new to this...Thanks
[image]

chillibasket, 2024-03-06:
@TheBanshee71 your error appears to be different. It says in the error message "failed at step USER" which means that the user name defined in the "walle.service" file is wrong. This is most likely due to using a different username than the default if "pi". Try updating the file with your actual username

TheBanshee71, 2024-03-06:
I found the problem, The username is pi but it didn't have correct permissions. But now I'm getting an cannot start audio device error.

TheBanshee71, 2024-03-06:
this is the only error im getting now
pygame.error: ALSA: Couldn't open audio device: Unknown error 524
Ive googled it with no luck. Any ideas? Thanks

dkrey, 2024-03-07:
Do you get audio if you start the webinterface manually as user pi?
If so, do the error happen again if you run the webinterface as root or via sudo?

If that's the case, you just need to add the env var to the unit file and reload it, just as I mentioned above.
`sudo nano /etc/systemd/system/walle.service` 
 Add this line below [Service]
`Environment=XDG_RUNTIME_DIR=/run/user/1000`

Reload the unit file
`sudo systemctl daemon-reload`

Restart the interface
`sudo systemctl restart walle`

Check the logs
`journalctl -u walle`

TheBanshee71, 2024-03-07:
I had added that before with no luck, but I re-installed everything and this time it worked!! Thanks

closed 2024-04-14 by chillibasket