chillibasket/walle-replica #30: Problem with video stream

Calst54, 2024-01-20:
Hi chillibasket
I built the wall-e with a raspberry4 and added the raspberry camera module 2. Raspberry is runnimng on OS 64bit. now everything motors, servos run fine on the web page. all files app.py, mjpgstreamer.sh are without modification except auto start for arduino and streamer. with the new OS the camera does not have to be enabled. that's automativ and the cameara is working when I run rpicam-hello -t- 0
my problem is that i have no video stream on the web server page. the auto start seems to work since i get the "end stream" message on the web page.
i tried several things but so far nothing seems to solve the problem. can you help?
as i am rather mechanical engineer I have no idea where to search...

chillibasket, 2024-01-21:
Hi! Thanks for letting me know that the camera is enable by default now, I'll update the instructions to mention that.

To double check that the video stream is actually running, you can open the stream it separately in your browser by using the the same IP address as the web-interface, but with ":8080" at the end instead of ":5000". If that is running, then the issue is with the web-interface that it is not loading this address correctly. If it is not running, most likely the configurations within the `~/mjpg-streamer.sh` file are incorrect for your camera.

Calst54, 2024-01-22:
[image]
Hello
Thank's for your fast response. I tried to run 192.168.0.168:8080 separate and in parallel to the :5000 but I get mo image. I tried to attach a picture of "my" wall-e in a military version. (olive, matte PLA) and it has the box of the camera in front of the robot. the camera is really the last step to have a fully working wall-e. I am anxious to get this working...
regards

chillibasket, 2024-01-22:
Great job on your Wall-E, I like the picture! The system uses "mjpg-streamer" in the background to get the video from the camera and show the live stream. It looks like the author of that tool hasn't updated it for the past 3 years, so it no longer runs properly on the newest RPi OS version (there were a lot of changes to how the camera is managed). It may be possible that the problem is fixed in one of the community forks of the library, for example in https://github.com/neuralassembly/mjpg-streamer/tree/master. There is also an issue in the library Github page which suggests they have found a potential workaround: https://github.com/jacksonliam/mjpg-streamer/issues/397

Unfortunately, I don't have the new raspberry pi to test different options with, so there is not too much I can do at the moment to solve the issue. If you manage to get the stream working by suggestions in the links above, let me know so that I can update the tutorial.

w45354, 2024-01-23:
yes, Flask is already installed in new RPI, you can check the list by typing sudo pip3 list. BTW, how do you find the serial port of arduino at the step 9a9B?

Calst54, 2024-01-23:
Indeed there is a whole list and Flask 2.2.2 is in it. 
As I am far away from a SW specialist I have no idea what step 9a9B is or how I get there. Is this some debug mode?

Calst54, 2024-01-23:
for a change. picure shows wall-e with his friends. two robots that I made with ESP32 CAM and one servo to turn the head.
[image]

w45354, 2024-01-24:
My USB camera works well. Maybe you can try another cam.

Calst54, 2024-01-25:
I have ordered a 2nd camera but i don't think it is a camera problem since i get cristal a clear picture with rpicam.
question: does any of those inteface settings needs to be enabled?
[image]

w45354, 2024-01-25:
I enabled SSH, I2C, and serial console

Calst54, 2024-01-26:
no success. also after doing a complete new installation I get everything working but no stream. As I am unable to find the problem i will abandon the project and have a blind walle!
Thanks for you patience.

w45354, 2024-01-27:
what a pity! your design is so great. would you mind if share your design?

dkrey, 2024-01-28:
I just got it working with a RPi v3 Camera, but it should work with a v2 camera just fine. 
My setup is a RPi 4 2GB running Raspi OS /Debian Bookwork / 12.4

Instead of mjpg streamer switch over to camera-streamer
https://github.com/ayufan/camera-streamer

The dpkg packages didn't work, but compiling it was straight forward. 

For the webinterface I only added a quick hack for the moment.
in app.py line 23  set `autoStartCamera = True`
in line 42 set `streaming = 1`

This should already work once you've got the systemd service of camera-streamer up and running.
As another precaution I commented out most of the code from the function `def onoff_streamer():`

Calst54, 2024-01-29:
@chillibasket: you probably like my new version. redesigned the door to fit a 4.3" display from waveshare and put the main switch in the front panel. I can stll use the HDMI port for a 2nd display. that works just perfect and looks very sophisticated. sure i will my files availabe but i still doing some design.
[image]
[image]

@dkrey : thanks for your comment. I have a second RPi with cam on the side where I can test and hook it to the andruino of walle.

Calst54, 2024-01-30:
@dkrey : when executing he system configuration raspi ask for a password (but not mine...) and then i get the error: failed to enable unit, 
any idea? 
[image]

Calst54, 2024-01-30:
i have no idea of the admin password. i took a new card and put the OS with imager on it. in the installation i set the PW to 1234 (its only a test card) and the host to walle. after that i installed the web server acc to the instructions form chillibasked and then made the changes you suggested. after this i used the link to the camera-streamer and this installs e file camera-streamer-generic_0.2.8.bookworm_armhf.deb and probably a lot of other files. then i used the recommended precompiled debian package for the installation. first part was ok and secon part for system configuration resulted in the error message.
but not mine...but not  
when i run the sudo as above i get: failed to enable unit: Unit file camera-streamer.service does not exist

when tried to used the direct way it can't finde the file camera-streamer. is probably in another path...

Calst54, 2024-01-30:
found the file in /bin and change the path but now i have the messsage the libcamera is not supported..

so far the only way to get a picture of my system ist with rpicamera-hello -t 0 
so i guess the RPi and the camera are working ok

dkrey, 2024-01-30:
Have uninstalled the dpkg package again? The libcamera message is exactly what I got when I used the .deb
Also most of the examples are missing from the deb packages.

Calst54, 2024-02-02:
would it not be better or simpler to use the raspberry command rpicam-vid which is directly streaming a video rather than use an external tool? 
with rpicam-vid -t 0 I have the video stream of the camera on the display of the RPi and this could be sent to the app.py
can you help?

chillibasket, 2024-02-02:
That may be an option. According to the [documentation](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/camera/rpicam_vid.adoc), the following command could be used to start a stream: `rpicam-vid -t 0 --inline --listen -o tcp://0.0.0.0:8080`. You can try running this command and then press the start webcam button on the web interface. If it works, then it should appear in the web interface video panel. 

If it works, the web interface script which is used to start the camera can be replaced with this simple command.

dkrey, 2024-02-03:
I gave it a quick shot, but the process only ran a second or two. But I'm keeping my fingers crossed that @Calst54 has more success.
In the meantime I stay with camera-streamer ;)

Here's my error log btw.
```
wall-e@wall-e:~ $ rpicam-vid -t 0 --inline --listen -o tcp://0.0.0.0:8080
[0:04:43.964998375] [2949]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+118-563cd78e
[0:04:44.017831887] [2955]  WARN RPiSdn sdn.cpp:39 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[0:04:44.020263149] [2955]  INFO RPI vc4.cpp:444 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media0 and ISP device /dev/media2
[0:04:44.020340928] [2955]  INFO RPI pipeline_base.cpp:1142 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
Preview window unavailable
Mode selection for 640:480:12:P
    SRGGB10_CSI2P,1536x864/0 - Score: 1486.67
    SRGGB10_CSI2P,2304x1296/0 - Score: 1786.67
    SRGGB10_CSI2P,4608x2592/0 - Score: 2686.67
[0:04:50.108312497] [2949]  INFO Camera camera.cpp:1183 configuring streams: (0) 640x480-YUV420 (1) 1536x864-SBGGR10_CSI2P
[0:04:50.109440858] [2955]  INFO RPI vc4.cpp:608 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
#0 (0.00 fps) exp 32876.00 ag 5.99 dg 1.01
#1 (30.01 fps) exp 32876.00 ag 5.99 dg 1.01
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to send data on socket
Aborted
```

RoBurr, 2024-02-04:
Hi I'm running in the same problem(s) camera stream does not work, As well autostart is not working with Raspi Pi 4 :-(

dkrey, 2024-02-04:
@RoBurr
Autostart most likely doesn't work because of a missing env variable.
Have a look at my comment here:
https://github.com/chillibasket/walle-replica/issues/29#issuecomment-1913716764

I never got mjpg-streamer running, but as noted above the raspi cam v3 works just fine with camera-streamer. You just need to compile it, since the packages are broken.

Calst54, 2024-02-09:
@chillibasket 
because i don't know where to put the string I tried it with the terminal and then opened Firefox and tried to access with :8080
on the RPi the preview window starts but on Browswer I see the following:
[image]

opening Chrome starts the preview window for a second ond then stops the process on the RPi

i assume some data is transmitted but in a format that the browser can't read

chillibasket, 2024-02-24:
Hi @Calst54, another Wall-E builder has come up with an installation guide using a different camera streaming library which looks promising here: https://wired.chillibasket.com/3d-printed-wall-e/#comment-1277

Since this comment thread has gone a little bit off topic I will be closing it and have opened a new Issue #32 which is aimed just at fixing the camera issue. However, feel free to continue adding comments to this closed thread if you do any more testing (if required, they can be referenced on the new Issue).

closed 2024-02-24 by chillibasket