chillibasket/walle-replica #31: sounds fail to play

w45354, 2024-01-25:
all of the sounds were OK to play at the beginning, but when I finished all the settings, the sounds could not be played anymore. what might be the issue?

dkrey, 2024-01-29:
A wild guess could be that your pi decided to use a different audio out than before.
Can you play the local .ogg files via aplay? 
You could also check amixer to set the volume and make sure nothing got muted.

w45354, 2024-02-02:
the sound issue may be due to this(shown in the picture). 
[image]
what does it mean by this error? also, with this error, I will meet problems when logging onto the web interface

dkrey, 2024-02-02:
Yeah, this error prevents the interface from working correctly. 
For some reason your raspberry has problems accessing the usb-serial port to the arduino. If you did change to the code, it might help resetting the git repo and start over.  In case you're fine with losing all changes to the code, enter  
`git reset --hard` in the walle-replica directory.
Afterwards you'd need to set the config in app.y again of course.

But maybe that's not even necessary. You could edit app.py, modify the variables arduinoPort and autoStartArduino:
`arduinoPort = "ttyACM0"`
`autoStartArduino = True`

If that also doesn't help, it might be interesting if you arduino is actually detected by the raspi. So next try this and see if the Arduino is listed there.
`lsusb`

w45354, 2024-02-03:
the interface issue is tackled. But sounds are still not played. And I am sure the sounds can be played straight away on my RPI, and it is not muted. When I play the sounds through the web interface, no error pops up.
[image]

w45354, 2024-02-03:
again, the sound is not working after I finish Automatically start Server on Boot setting. Maybe something happening during the camera stream phase or auto-start phase causes the sound issue?

dkrey, 2024-02-03:
Does it work again when you manually start the webserver?

Also what's the output of journalctl right after you log in the webinterface?
`journalctl -u wall`

w45354, 2024-02-04:
still not working when manually starting. the journal is like this.
[image]
[image]
[image]
[image]

dkrey, 2024-02-04:
Is this a recent 64bit raspberry pi os you're using? At least the original systemd unit file should throw this error `pygame.error: No available audio device`. Also mjpg streamer seems to be running for you (which is good but not common 😄  )

For the second test I meant: what happens if you start the interface in user context. Do you get sound again?
like `pi@walle$ python3 ~/walle-replica/web_interface/app.py`

And a last question, can you play audio in terminal? 
´pw-play ~/walle-replica/web_interface/static/sounds/Sound_Ohhh_1850.ogg`

(replace pw-play with something like ogg123, if you don't have it installed)

w45354, 2024-02-06:
it's so weird, sometimes work sometimes not. I guess, if any possibility that the ports of the speaker and camera conflict. as my speaker is a little special, it has a jack output and a USB power supply. and also, my camera is a USB one rather official one. maybe changing hardware would tackle this issue.

dkrey, 2024-02-07:
Hmm - could it be that the speaker draws too much power from your pi? Or does it add another sounddevice via usb so the pi doesn't know which one to use? You use `alsamixer` via terminal to check if there's anything muted or what the current  audio outputdevice is.

closed 2024-02-26 by w45354