chillibasket/walle-replica #32: Video stream no longer works with the newest RPi OS

chillibasket, 2024-02-24:
The old video stream method using `mjpeg-streamer` no longer works on the newest RPi OS and the project appears to no longer be receiving any new updates. This streamer needs to be replaced with a different library, for example using the PiCamera2 project: https://github.com/raspberrypi/picamera2

A provisional step-by-step installation guide has been suggested by Vulterey on the Wall-E build guide page: https://wired.chillibasket.com/3d-printed-wall-e/#comment-1277

It may be a little while before I can test this procedure and integrate these changes into the repo, so if anyone can test it and confirm that the process works, this would be a great help!

Calst54, 2024-02-25:
Thank vou for the information. I followed the instructions of vulterey and got the streamer working perfectly. The procedure is simple and straight forward.  To get a larger image I changed the size to 800 x 600 and this works as well.
I tested the wall-e form iPhone, computer and tablet and got a clear picture on all devices while having wall-e driving around.
Thanks for to support and the great design.

chillibasket, 2024-06-09:
This feature has now been added in the latest release v3.0: https://github.com/chillibasket/walle-replica/tree/v3.0

closed 2024-06-09 by chillibasket