chillibasket/walle-replica #33: Features and Changes

dkrey, 2024-02-28:
Hi Simon,
I've made quite a few changes to the code -maybe a bit too much for a single pull request. So I wonder if you're interested in merging all of this anyway.
Things I did:

- Switched from pygame to alsamixer (changing the volume only works with linux)
- renamed app.py to web_interface.py so you can keep both versions
- added a config for uwsgi
- added local_config.py
- removed debug printing
- refreshed the libraries to the newest minor release
- removed mjpg streamer support and switched over to camera streamer
- added (offline) text-to-speech via espeak-ng and rubberband
- added the ability to script Wall-E's movements via code blocks.

At least my kids have lots of fun learning to code with blockly and Wall-E. I don't think they ever asked me before what a for-loop does 😁 
Here's a screenshot of the new code tab:
[image]

And here's my fork:
https://github.com/dkrey/walle-replica

chillibasket, 2024-02-29:
Hi! Those are some amazing improvements and definitely would be cool to add into the main project. It's great to see other makers building on the project and bringing it to the next level! Do you have a video of the robot in action with these extra features?

I'll have to have a think about what impact these changes will have on the build and setup instructions. Maybe it would also be good to have an easy way of turning on/off features during the installation (text-to-speech, code blocks) depending on user preferences. 

I'd say you can start by creating a pull-request into the "development" branch of my main repo, and we can discuss improvements, testing and integration there. Which Raspberry Pi and which OS are you using?

dkrey, 2024-03-01:
Thanks 😁 I'll see if I can get a video ready this weekend. 

The code blocks are just plain javascript and don't have any external dependencies (except the ones included in the lib folder). They just execute ajax api calls, like the rest of the gui.

Text-to-speech depends on espeak-ng which is included in the Debian repos. To raise the pitch of the voice, I've used rubberband (also in the repos), but that one is already optional. 

My Wall-E build runs on a Raspberry Pi 4 2GB, the OS is the most recent 64bit Raspberry OS (Bookworm).

The only annoying part is that the camera-streamer packages are broken at the moment, so camera-streamer needs to be compiled manually. But hopefully that'll change again. Anyway, I'll see I can clean up the code some more and open a PR.

chillibasket, 2024-03-01:
Great! It might make sense to change to PiCamera2 anyway as discussed in #32. Some provisional installation instructions were shared by Vulterey [on my website](https://wired.chillibasket.com/3d-printed-wall-e/#comment-1277) and he might make a separate pull request for that.

dkrey, 2024-03-04:
The video is finally up 🎉 
https://youtu.be/yW8i79MgR_8

PiCamera2 looks very promising, although the mjpeg_server code appears to be more like a proof of concept. 
As mentioned, apart from the packaging issue, camera-streamer runs very well and stable.

chillibasket, 2024-04-14:
Thanks again for sharing these great updates! I have now merged them into the development branch #34 and will begin work on updating the instructions and making some minor refinements. My goal for release on the master branch is at the start of May.

dkrey, 2024-04-17:
Thanks a lot for merging these changes, I'm very much looking forward to the new release.

chillibasket, 2024-06-03:
Hi @dkrey,

I finally have almost everything ready for all the new changes to be released on the master branch. I included quite a few additional improvements and am putting together a script to automatically complete all the installation steps.

One issue I came across is that the audio will not play when the web interface is started as a service. The command "pw-play" gives an error. Did you need to do anything special to get the interface to start on boot and still be able to play sound on your system?

dkrey, 2024-06-03:
Hi @chillibasket 
great news. I'm looking very forward to update to the latest stable myself. 

Regarding the sound issue: as soon as I removed all pygame references and replaced them with their cli counterparts, it worked fine for me.

```pw-play``` is the pipewire player, that came with my distro (the current raspberry pi os, based on bookworm / Debian 12)

It should also be safe to use ```aplay``` instead.

chillibasket, 2024-06-09:
Thanks for the info. In the end I used aplay but I think I was just missing the environment setting in the service file to give it access to the audio subsystem. I have now finished updating the instructions and released everything into the main branch.