chillibasket/walle-replica #35: Newbie trying to incorporate Wall-E into my 3D Printing Course...

ScottPeterson2, 2024-03-07:
Hi, I have had my students 3D print all of the pieces of the robot.  I have purchased all of the electronics for the project using the list you provided.  I have connected all of the electronics using the diagrams you provided.  When I have my students open the wall-e arduino sketch and upload it to the Arduino uno we are only able to get the gear motors to work using the commands you mentioned on your website.  Unfortunately, we can not get any of the servo motors to move or make noise.  Do you have any input for what we should be checking.  We have installed the Adafruit PWM servo driver library 3.0.1.  We are powering the arduino from USB to computer and a 12volt battery pack connected to the buck board and motor shield.  I have no previous experience with Arduino, but I am a quick learner.  If you have any thoughts that might help us move forward I would greatly appreciate it.

[image]

[image]

[image]

[image]

[image]

ScottPeterson2, 2024-03-07:
I found a sketch that would allow me to connect a servo pinned out to the arduino.  The sketch operated the servo, so I know the servos are good.  I just ordered two more pca9685 servo controller boards.  I am thinking that the board that I received was bad.  I tested voltage at the buck board input and was about 12 volts and then checked voltage leaving the buck board and was roughly 5volts.  That board seems to be fine.  I am thinking that somehow signal is not either being sent or received by the servo controller.

chillibasket, 2024-03-10:
I can't see any obvious issues in the images. You could try running the example code which comes with the PWM servo driver library to see if that works: `File > Examples > Adafruit PWM Servo Driver Library > Servo` 

Also you can go through the Adafruit tutorial for the servo driver board to see if that helps you spot the issue: https://learn.adafruit.com/16-channel-pwm-servo-driver/overview

closed 2024-06-09 by chillibasket