chillibasket/walle-replica #37: Web interface problem

Nuno14LX, 2024-04-30:
Good morning, 
I’m printing Wall-e and testing the electronics in the meantime. I can’t launch the web interface after the password. I’ve tried everything that’s written in a post close but trien does it.
attached the terminal display.

[image]
Thank you for your help.

chillibasket, 2024-04-30:
The error says it cannot find the sound files. This is because your account username is "walle" rather than the default of "pi". Therefore you need to update the path to the sound folder in the "app.py" configurations [line 29](https://github.com/chillibasket/walle-replica/blob/95d4019bf357b51161a366f74aaee5464d69478b/web_interface/app.py#L29) to use "walle" instead of "pi"

Nuno14LX, 2024-04-30:
Good evening,

Thank you for your answer. She solved my problem.
And a big thank you for your work.

closed 2024-05-01 by chillibasket